FabMo/FabMo-G2-Core #15: Small Arcs Fail as segments instead of arcs, at speed

tedh-shopbot, 2021-11-08:
This is a problem seen in sample cutting files from Bill Young. In these files, spiral plunges are used to drill a hole with a terminal flat full circle (made from 4 arcs) executed at the bottom of the hole.

The problem can be duplicated with just a positioning move followed by the 4 arcs of the circle. 
G1 X28.90260 Y20.52364 Z-0.485 F360
G2 X28.9533 Y20.45124 I-0.01085 J-0.06155
G2 X28.88089 Y20.40054 I-0.06155 J0.01086
G2 X28.83019 Y20.47294 I0.01085 J0.06155
G2 X28.90260 Y20.52364 I0.06155 J-0.01085

When this is done at 6inches/sec as above(F360), then the moves are just a 4-diagonals in a diamond pattern. When the move is done at .25inches/sec (F15), then the moves are done correctly as a smooth circle. Something about the jerk planning here, rather than just doing the circle at an appropriately slow speed, is causing the motion to be expressed as (relatively fast) segments. NOTE that these are small segments.

RobMackie, 2021-12-29:
Merged in pull request #21

closed 2021-12-29 by RobMackie

wlyoung, 2022-01-11:
This is most likely the same issue, but I mostly saw this happen in small
circles but not always. On occasion it would truncate things like a
radiused corner on a larger rectangle, moving from the end of one straight
segment to the beginning of the next one in a straight cut and skip the
radiused corner

-- 
Bill Young

www.shopbottools.com
***@***.***/
www.shelter20.com
www.100kgarages.com
www.instagram.com/seaside5592/

Cell: (757) 999-5592

"There are hundreds of thousands of problems around the world that fall
into the category of 'having the right shape in the right material in the
right place at the right time'. -- Andreas Bastian

tedh-shopbot, 2022-01-11:
Hey Bill, if you have an example I'll test it ... otherwise we'll wait to see how the fix works for you.