FabMo/FabMo-G2-Core #20: Spindle dwell is not happening on resume after pause.

RobMackie, 2021-12-14:
During at a Resume with the spindle being used, it looks like the pause is made, then once resume is started, the spindle is turned on and the plunge happens. It should be: Spindle on, then Pause, then Plunge (the point of the pause is to allow the spindle to spin-up). I have reported previously but not sure on the g2 list.

In a fix, the dwell after activating the spindle will happen before motion starts.

tedh-shopbot, 2021-12-31:
Just copying in relevant thread from discussion on Synthetos slack channel:

TED:

Hey @Bliss Brass . I've compiled your fix and I'm about to go try it ...
However, I'm a little worried that we all have gotten a little confused on this ... and I may have helped create the problem recently reporting incorrectly to @Rob Mackie .
Let me start with a little history. Here is where we were after accepting the Pull Request on "Item 24: Spindle on Off issue#12"
From me:
_"I believe this change fixes the primary problem intended for ITEM 24, which was the fact that if there was an M5 (to turn off the spindle) in a file after the spindle was on; that if there was a feedhold done before the M5, that the spindle would not turn back on after a "resume" (when the file was sufficiently short). That seems to be fixed -- as tested with the original demo code, and a more complex file.
From a minor DISPLAY perspective, the one thing I notice is that while the #1 output is reported immediately when the spindle is turned on with M3; when the hold is resumed from, the update of the #1 output status does not occur until after the dwell and plunge back into the material have occurred (this can be seen in the G2 output). If there were a simple way to do it, a report when it actually turns on would be best.
Perhaps more problematic (and something I had not noticed) is that when the spindle is turned off for the feedhold, it is turned off before the pull up. This might create a situation where the cutter binds in the material as it is being removed. I believe, and we want RobM's input here, that if we could accomplish it, we would like the spindle to turn off after it has been retracted."_

So at that point, we had the display issue such that when at a feed-hold in which the spindle was being retracted and then resumed from, the spindle correctly turned on immediately at the "resume", then paused (dwelled), and then re-plunged to the previous depth. That is, the behavior of the Z and the spindle was correct. However, the sr report of the #1 or spindle being turned back on was delayed until after the Z plunge. This gave the impression that the output was not turned on until after the dwell, even though it was on OK earlier. So a "fix" for this issue would just be a change in the timing of the display, via an earlier sr report of the output (or spindle) having been turned on.

I'm a worried that Rob and I may gotten a little sloppy, confused ourselves, and started believing the output indicator (because we weren't actually running tools) ... and gotten everyone else confused ??? And, thus the bug as issue #20 has come to be reported slightly different than it is???

So back to where we were, and still are, I think. Beyond the display issue, we have the problem that when the feed hold is initiated, the first thing that is happening is the spindle turns off, and then the Z is retracted. It should be that the Z is retracted and then the spindle turned off. No dwell involved here.

The overall sequence of behavior on such a normal, in-motion, feed-hold INITIATION should be:
**Feed-Hold requested  >  motion stops  >  z-retracted (per option setting)  >  spindle turned off and reported**

A RESUME from Feed-Hold should be:
**Resume requested  >  spindle turned on and reported (per optionality)  >  spin-up dwell applied  >  z re-inserted  >  motion resumed**

OK. I'm going to paste this into the issue ... and go see what your current version does :slightly_smiling_face:

RobMackie, 2022-01-04:
New bugs #24 and #25 opened to deal with issues encountered as we discovered Rob M had written up a junk bug.

closed 2022-01-04 by RobMackie