FabMo/FabMo-G2-Core #31: Hardware feed hold outside program execution

kyle-kingsbury, 2022-01-25:
Feed hold can be toggled even when a cnc program is not executing. This is not inherently wrong, but deviates from the FabMo expectations. If the button is pressed outside program execution is should be ignored.

kyle-kingsbury, 2022-01-25:
In cycle_feedhold.cpp, moved

// Reset the request if it's invalid
	if ((cm1.machine_state != MACHINE_CYCLE) || (cm1.motion_state == MOTION_STOP)) {
		cm->hold_state = FEEDHOLD_OFF;          // cannot honor the feedhold request. reset it
	}
To the top of
 
void cm_request_feedhold(cmFeedholdType type, cmFeedholdExit exit)

and encased the rest of the function in an else statement. So if the state is not a machining cycle, or if motion is already stopped, the button will be ignored.

RobMackie, 2022-01-28:
Closed with merge of PR #33  Will need to open new issue to deal with feed hold during spindle spin up after feedhold but that will likely be in fabmo

closed 2022-01-28 by RobMackie