FabMo/FabMo-G2-Core #67: Instrumentation for digital analyzer should be behind preprocessor guards run included from make command line

RobMackie, 2022-11-02:
Currently, the build includes instrumentation that interferes with the normal operation of axis A&B (a.k.a. stepper driver channels 4 & 5).  

Channels 4 and 5 cannot be used on main at this time. We never want to ship with this instrumentation turned on, so it should be based on a #define in the code but should always be defined on the Make command line.

The appropriate command line for including the instrumentation looks like this:
make CONFIG=sbv300  USER_DEFINES=INSTRUMENT_SEGMENTS_N_MOVES

without USER_DEFINES=INSTRUMENT_SEGMENTS_N_MOVES, the instrumentation will be preprocessed out of the file.

tedh-shopbot, 2022-11-14:
Just noticed that for the PR#68 (guarding) there are 3 more instances of motor_.5.stepStart that are being called and not included yet in the "processor guards on instrumentation".

There are 2 in cycle_feedhold.cpp and 1 in plan_exec.cpp.  The latter that is the most interesting because it gives us a mark when the upcoming segment has been generated and then we get the second mark at the instant that it is loaded into the stepping runtime.

The ones in cycle_feedhold just allow checking to see that the block is correctly defined in the secondary planning that is used for the pull-up and put-down.

RobMackie, 2022-11-17:
Ran the normal dev checks on the vanilla build. Built it with the flag in place for the instrumented build. Built cleanly. Did not put a scope on it for the instrumented build to catch the 5th axis. Did run the preprocessor and inspect the output for cycle_feedhold.cpp and plan_exec.cpp. It is correctly present and the  build succeeds. seems sufficient proof since we've been using this for weeks. 
Kyle if you feel I need to do that let me know. I left my scope in the office so can't check it until I get back in.

RobMackie, 2022-11-30:
closed as of Pull Request #73

closed 2022-11-30 by RobMackie